pablogventura/OctoPrint-PowerFailure #1: [HELP] Resume height problem

Rabbit-404, 2018-02-24:
Hi, im having a problem, i was testing the plugin, and everytime I cut the power off it restarts printing mid air, how can i solve this?

pablogventura, 2018-02-25:
Hi! Thank you for testing it! :-) Too bad it doesn't work out for you, though. :'( What kind of printer do you have? The z-axis moves the bed or the nozzle? There are printers where the lack of power causes the bed to go down by gravity, and that could only be solved by using a endstop for the zmin, although I have never tried it.

Rabbit-404, 2018-02-26:
its a I3 clone, the z axiz moves the nozzle

pablogventura, 2018-02-26:
Do you have the Z_HOMING_HEIGHT activated in the marlin? If so, you must
also configure it in the options of this plugin.

Rabbit-404, 2018-02-28:
That was it, now it starts at the correct height, but to the left side of the print =(

pablogventura, 2018-02-28:
Can you send a video?

Rabbit-404, 2018-02-28:
I uploaded it to Google Drive, here is the link
https://drive.google.com/file/d/1LCUSfR-Wd5HlEDT4gh33ZHQ4xqgjNJ0D/view?usp=sharing

Rabbit-404, 2018-04-10:
Im guessing there is no workaround for this yet =(

pablogventura, 2018-04-10:
I'm sorry! I've been very busy and now I'm moving to another apartment. For
now I won't be able to continue with this development, but when I have more
free time I will continue with this. Any pull request is welcome! Greetings!

pablogventura, 2018-04-10:
I just saw the video. It seems to me that the start code of your gcode
(sliced to be used in your machine) must be different from the configured
in the plugin.
Surely if you check your gcode startup code and the one in the plugin
configuration you can fix this.