pablogventura/OctoPrint-PowerFailure #5: Is it works with klipper?.

anand-010, 2023-01-15:
I am using SKR PICO with klipper firmware.
I 've installed this plugin on octoprint.
I've started a print and and after sometime power off the printer both raspberry pi and skr pico.
then i turned on both the mcu automatically  connected to the octoprint but the print is not resuming.
Anything wrong with my setup ?.

pablogventura, 2023-01-15:
I have only tested it in Marlin, it may not work in Klipper. :-(

paukstelis, 2023-01-15:
Klipper typically requires Z axis homing before it will allow commands to be passed. There may be a way to turn that off.

paukstelis, 2023-01-15:
I will try to add some Klipper support.
In order for it to work your Klipper config must have a `[force_move]` section with enable_force_move=true
Your XY gcode section should include `SET_KINEMATIC_POSITION x=50 y=50 z={currentZ}` at the top.

paukstelis, 2023-01-15:
I have created a klipper branch in my fork. However, I have discovered that Klipper behaves oddly. If the current z height is less than the Z_HOMING_HEIGHT equivalent, it will lift on a G28 X Y, but if it is above that  value, it will not lift. This means there will have to be some conditional stuff put in.

paukstelis, 2023-01-23:
> Please try [force_move] enable_force_move: True and SET_KINEMATIC_POSITION X=0 Y=0 Z=0

Klipper support has been added in the lastest release.

anand-010, 2023-01-27:
I've tested it today but I've intentionally power off during a print.
after turned on the printer automatically connected to the printer but no sign of print resuming or a notification.
the extruders are not even try to pre heat.
I've setted [force_move]
enable_force_move: True
and not setted [safe_z_home].
Is there any logs that i can check?.

anand-010, 2023-01-27:
It worked after a try thank you.

anand-010, 2023-01-27:
It only works on first power loss second power loss it doesn't automatically started even after start printing recovery file it's old position print on new height.

anand-010, 2023-01-27:
after a long tests I've found that it's not about second power loss.
It's sometimes works but sometimes don't.
is it currepts the json file powerfailure_recovery.json when writing to it?.
in that case we want two files for fail proof run.

paukstelis, 2023-01-27:
SD cards that experience many power losses may certainly get corrupt, that is pretty well known with RPis. I only pulled the power once on my testing unit. If  you want to test if that is really the issue, you can simply disconnect the printer during a print, then reconnect. That will trigger all the same things in the plugin without the need to turn off the power.

markus-nclose, 2023-01-31:
Hi adding to the tread, I'm running Ender 3 s1 with OctoKlipper. I'm struggling to get the job to resume. 

Steps: I start a job, monitor the files (see below), and remove the USB connection. I then restore the USB connection and connect klipper to the printer again, but either firmware restart OR connect button.

The `print_recovery_data.yaml` gets created (not sure if is part of this plugin), but not any other GCODE files and nothing further happens. 

I live in RSA with a power failures every 6 hours, it would be great to get this working :) 

```
}pi@raspberrypi:~/.octoprint/data/powerfailure $ cat ./powerfailure_recovery.json
{
    "bedT": 60.0,
    "tool0T": 210.0,
    "filepos": 15648,
    "filename": "Sir_Flexi_Raptor_2h12m_0.20mm_205C_PLA_ENDER3S1.gcode",
    "currentZ": 0.2,
    "last_X": 37.764,
    "last_Y": 130.041,
    "recovery": true,
    "powerloss": true,
    "extruder": 0.00452,
    "extrusion": "M83",
    "feedrate": 1200.0,
    "last_fan": "M107",
    "linear_advance": null
}pi@raspberrypi:~/.octoprint/data/powerfailure $ cat ./powerfailure_recovery.json
{
    "bedT": 0,
    "tool0T": 0,
    "filepos": 0,
    "filename": null,
    "currentZ": 0,
    "last_X": 0,
    "last_Y": 0,
    "recovery": false,
    "powerloss": false,
    "extruder": null,
    "extrusion": null,
    "feedrate": null,
    "last_fan": null,
    "linear_advance": null
}pi@raspberrypi:~/.octoprint/data/powerfailure $ ls ../
announcements  appkeys  backup  klipper  plugin_blacklist.yaml  pluginmanager  powerfailure  preemptive_cache_config.yaml  print_recovery_data.yaml  softwareupdate
pi@raspberrypi:~/.octoprint/data/powerfailure $ cat ../print_recovery_data.yaml
date: 1675086643.067723
origin: local
path: Sir_Flexi_Raptor_2h12m_0.20mm_205C_PLA_ENDER3S1.gcode
pos: 22317
```

paukstelis, 2023-01-31:
print_recovery_data.yaml is not relevant.

The powerfailure_recovery.json file is the important one.
I do not know when you are using `cat` on that file, but the first one shows it is working. The second one is after the file has been 'cleaned' which happens after a recovery file is generated or after a print is finished or cancelled.

anand-010, 2023-01-31:
I've found a method to always get correct data without corruption.
we need two files like powerfailure_recovery.json.
We only want to update it in a counting manner (if first second data is stored on one file store the next second on the other file and continue like that).
in that way if the lastest file is corrupted we can use the previous one using the method and the don't need additional writing to the sd card.
is it appliable?.

markusza, 2023-01-31:
> print_recovery_data.yaml is not relevant.
>
> The powerfailure_recovery.json file is the important one. I do not know when you are using `cat` on that file, but the first one shows it is working. The [...]

Okay thanks for clearing it up.
I do a `watch cat ./*` on the folder, the second one is created after the power failed. Not sure what to troubleshoot next or how?

paukstelis, 2023-01-31:
You can turn on plugin debugging which provides a little more information, but  that file will only switch to null value after the print is completed, cancelled, or the recovery file is generated. Are you sure you dont have a file called `recovery_<myfilename>` somewhere?

markusza, 2023-01-31:
Tested again now, almost immediately when I plug the USB cable out, the NULL value file is created. 

```
tail: powerfailure_recovery.json: file truncated
{
    "bedT": 60.0,
    "tool0T": 205.0,
    "filepos": null,
    "filename": "Sir_Flexi_Raptor_2h12m_0.20mm_205C_PLA_ENDER3S1.gcode",
    "currentZ": null,
    "last_X": 99.452,
    "last_Y": 116.554,
    "recovery": true,
    "powerloss": true,
    "extruder": 0.10116,
    "extrusion": "M83",
    "feedrate": 1500.0,
    "last_fan": "M106 S255",
    "linear_advance": null
}tail: powerfailure_recovery.json: file truncated
{
    "bedT": 0,
    "tool0T": 0,
    "filepos": 0,
    "filename": null,
    "currentZ": 0,
    "last_X": 0,
    "last_Y": 0,
    "recovery": false,
    "powerloss": false,
    "extruder": null,
    "extrusion": null,
    "feedrate": null,
    "last_fan": null,
    "linear_advance": null
}tail: powerfailure_recovery.json: file truncated
```
No file named `recovery_X` 

```
pi@raspberrypi:~/.octoprint/data $ find ~/ -name recovery*
/home/pi/oprint/lib/python3.7/site-packages/octoprint/static/css/recovery.css
/home/pi/oprint/lib/python3.7/site-packages/octoprint/static/js/recovery
/home/pi/oprint/lib/python3.7/site-packages/octoprint/static/js/recovery/recovery.js
/home/pi/oprint/lib/python3.7/site-packages/octoprint/static/less/recovery.less
/home/pi/oprint/lib/python3.7/site-packages/octoprint/templates/recovery.jinja2
pi@raspberrypi:~/.octoprint/data $
```

paukstelis, 2023-01-31:
Also, provide contents of octoprint.log. If there are errors they should show up there.

markusza, 2023-01-31:
[octoprint.log](https://github.com/pablogventura/OctoPrint-PowerFailure/files/10547767/octoprint.log)

For some reason there is no debug log created for the plugin which seems weird. 
[image]

markusza, 2023-01-31:
```
2023-01-31 15:02:14,245 - octoprint.plugins.powerfailure - DEBUG - Connected Event. Check Recovery
2023-01-31 15:02:14,349 - octoprint.plugins.powerfailure - DEBUG - Checking recovery
2023-01-31 15:02:14,378 - octoprint.plugins.powerfailure - DEBUG - There was no print failure.
2023-01-31 15:03:31,188 - octoprint.plugins.powerfailure - DEBUG - Timer condition met
2023-01-31 15:03:31,189 - octoprint.plugins.powerfailure - DEBUG - Timer cancelled
2023-01-31 15:03:31,189 - octoprint.plugins.powerfailure - DEBUG - Timer finished
2023-01-31 15:03:31,206 - octoprint.plugins.powerfailure - DEBUG - Timer started
2023-01-31 15:07:12,680 - octoprint.plugins.powerfailure - INFO - PowerFailure: Print failed with cancelled
```
Found the  debug logs in octoprint.log file. Okay looks like klipper sees it as a cancelled job and PowerFailure plugin doesn't resume it for that reason. Mmh.

```
2023-01-31 15:07:12,375 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
```

paukstelis, 2023-01-31:
Yeah, that is different from the behavior on my klipper setup. Will have to explore what is leading to the cancel event.

paukstelis, 2023-01-31:
What do you have for these settings?
[image]

markusza, 2023-01-31:
I don't have power now lol, #SouthAfrica so will check tomorrow for you. If it helps, it is still a default Octoklipper install

markus-nclose, 2023-02-01:
You are correct, I have changed the default setting to match yours above and it has created the file. 
Thanks for assisting and thanks for pickup up the development on this again.

paukstelis, 2023-02-01:
> I've found a method to always get correct data without corruption. we need two files like powerfailure_recovery.json. We only want to update it in a counting manner (if first second (time t) data is stored on one file store the next second(time t) on the other file and continue third second (time t) on first file like that). in that way if the latest file is corrupted we can use the previous one using the method and the don't need additional writing to the sd card. is it appliable?. we can also introduce a counter in the json to get latest one. totally there is two files.

I think this is probably premature. Unless this corruption issue is something others see as well. This could be related entirely to the physical SD card you are using.

markus-nclose, 2023-02-02:
> I think this is probably premature. Unless this corruption issue is something others see as well. This could be related entirely to the physical SD card you are using.

That being said, I was running raspberry of USB drive instead of SD card and noticed something similar. The file is truncated when it is updated with new parameters, I suspect my one test powerfailure happened during this truncation event and the file was blank. I wonder if  @anand-010 is not experiencing this?

You could see this in my log further up the thread. 
```tail: powerfailure_recovery.json: file truncated```

paukstelis, 2023-02-02:
> That being said, I was running raspberry of USB drive instead of SD card and noticed something similar. The file is truncated when it is updated with new parameters, I suspect my one test [...]

I think that has to do with the tail command. The output you show is not truncated, so I do not think that is the case.

Hairyloon, 2023-07-01:
> Klipper typically requires Z axis homing before it will allow commands to be passed. There may be a way to turn that off.

I don't know if you can turn it on, but you can use [safe_z_home], which you can set to move Z up, then over to a safe place before homing.

paukstelis, 2023-07-01:
> > Klipper typically requires Z axis homing before it will allow commands to be passed. There may be a way to turn that off.
> 
> I don't know if you can turn it on, but you can use [safe_z_home], which you can set to move Z up, then over to a safe place before homing.

Directly from the README: 
Klipper firmware. You must have the [force_move] section with the enable_force_move=true option in your Klipper configuration. Check the appropriate box in the settings. If [safe_z_home] is set, use the z_hop value as Z_HOMING_HEIGHT.

closed 2023-07-01 by paukstelis