pablogventura/OctoPrint-PowerFailure #10: Feature Request: Configurable write location

Boothin, 2023-01-28:
Since this plugin will make so many writes to disk, I think the ability to configure the location of the backup file(s) would be great. This would allow the use of ramdisk (if the octoprint host is on a UPS battery backup) and write the file from ramdisk to a persistent location if the battery backup triggers.

paukstelis, 2023-01-28:
I agree this would be nice, but this may be outside the scope of what a plugin has the ability to do. One could (presumably) just symlink the ~/.octopriont/data/powerfailure directory to the ramdisk location.

Iron3eagle, 2023-03-03:
@paukstelis, this would be a helpful feature if possible.  I had my first single second power outage that stopped a print, but I can't seem to locate the generated resume GCODE.  When I SSH in and check the location you gave (I assumed this is where it would be) there is only a single JSON file that I assume is the template for building the file.  Where should this GCODE file exist?

paukstelis, 2023-03-03:
> @paukstelis, this would be a helpful feature if possible. I had my first single second power outage that stopped a print, but I can't seem to locate the generated resume GCODE. When I SSH in and check the location you gave (I assumed this is where it would be) there is only a single JSON file that I assume is the template for building the file. Where should this GCODE file exist? I also check ~/oprint/lib/python3.7/site-packages/Power_Failure_Recovery-1.1.1.dist-info but couldn't seem to find anything there either.

The only place the plugin writes is ~/.octoprint/data/powerfailure and based on how OctoPrint plugins work, that is the only place where plugin data should be written. The recovery gcode is generated and placed in the ~/.octoprint/uploads/. The recovery file is not generated until the printer is reconnected to the server.

Iron3eagle, 2023-03-03:
@paukstelis thanks, that's the default upload location I see, I'm guessing the power outage might have been too quick to trigger the plugin then, or it was never writing to the temp file.  Let me know if you need any debug files I can provide.

paukstelis, 2023-03-03:
For starters just verify that the json file is being written as a print is on-going. Start a print, ssh into device, and just check the file contents. You should see it changing.

paukstelis, 2023-03-03:
Also, make sure under Printer settings that error handling is set to Disconnect and not Cancel Print. Cancelling the print will reset the file tracking.
[image]