pablogventura/OctoPrint-PowerFailure #11: After a restart of the server not work

mattcalavera, 2023-01-28:
The octoprint server stop during a print.
In the directory where the plugin save position there was correctly last state saved as a json file :
~/.octoprint/data/powerfailure $ cat powerfailure_recovery.json 
{
    "bedT": 70.0,
    "tool0T": 235.0,
    "filepos": 29313854,
    "filename": "PETG_SILICA50NOCORE_x3.gcode",
    "currentZ": 36.05,
    "last_X": 70.845,
    "last_Y": 100.787,
    "recovery": true,
    "powerloss": true,
    "extruder": 204.99197,
    "extrusion": "M82",
    "feedrate": 2442.6,
    "last_fan": "M106 S255",
    "linear_advance": null

But next mornign when i restart server I cannot see no way to get the printer resume the work!
How it is supposed to happen?

paukstelis, 2023-01-28:
The recovery file will be generated once the printer is reconnected to the server.

mattcalavera, 2023-01-28:
So it is supposed to work by generating a new file, which we have to manually print? What’s the name of the new file?

paukstelis, 2023-01-28:
The file name is `recovery_thefileyouwereprinting.gcode`. Take a look at the plugin settings. It describes everything. You also need to check if you have to set Z_HOMING_HEIGHT.

mattcalavera, 2023-01-28:
Ok thanks

mattcalavera, 2023-01-29:
excuse me, but I cannot find file....i search all over the file sistem... where is supposed to save file?

mattcalavera, 2023-01-29:
I try more metodically today. I started a simply print (an oval test), and after a while layers unplug pi power :
*) Immediately printer stops
*) I plug in power
*) octoprint restart

* )I enter via terminal and look for recovery file : 
XX@XXXXXX:~/.octoprint/data/powerfailure $ cat powerfailure_recovery.json 
XX@XXXXXX:~/.octoprint/data/powerfailure $ 
Empty file

* )I enter via terminal and search for a file name named r* o R* : 

XX@XXXXXX:/ $ sudo find -name R*.gcode
./home/XX/.octoprint/uploads/FLSUN SR/TESTs/retraction/Retract_PETG_2mm.gcode
./home/XX/.octoprint/uploads/FLSUN SR/TESTs/retraction/Retract_PLA_2mm.gcode
./home/XX/.octoprint/uploads/FLSUN SR/TESTs/retraction/Retract_PLA_220_3mm.gcode
./home/XX/.octoprint/uploads/FLSUN SR/TESTs/retraction/Retract_PLA_1mm.gcode
./home/XX/.octoprint/uploads/FLSUN SR/TESTs/retraction/Retract_PETG_3mm.gcode
./home/XX/.octoprint/uploads/FLSUN SR/TESTs/retraction/Retract_PETG_1mm.gcode
./home/XX/.octoprint/uploads/FLSUN SR/TESTs/retraction/Retract_PLA_3mm.gcode
XX@XXXXXX:/ $ sudo find -name r*.gcode
XX@XXXXXX:/ $ 

No recovery file

why?

paukstelis, 2023-01-29:
Let's start with simple example.
1. Begin a print.
2. Verify that you have a recovery json file is being generated: `cat ~/.octoprint/data/powerfailure/powerfailure_recovery.json`
3. Unplug USB from the Pi. The print will stop and printer will disconnect.
4. Reconnect printer in OctoPrint UI interface. At this point a recovery file should be generated.

mattcalavera, 2023-02-05:
Yes, I confirm : the script work

closed 2023-03-14 by paukstelis