pablogventura/OctoPrint-PowerFailure #12: Z height correction for recovery

camarade35, 2023-01-29:
Hello,
I voluntarily disconnected the cable to test the plugion. The printer returned home but when I ran the recovery file, the print went back too high.
Is there a way to adjust the pickup pitch or just the Z-hop pitch when home?
I don't use Klipper for information

I see in the Gcode this

; Klipper Z is not checked
; Z enable is not checked

G90 ;absolute positioning
G28 X0 Y0 ;home X/Y to min endstops
G92 Z3.1 ;set Z with any homing offsets
;M211 S0 ; Deactive software endstops
G91 ;relative positioning
G1 Z-0.0 F200 ; correcting Z_HOMING_HEIGHT

Excuse my english, that's translate from french

paukstelis, 2023-01-29:
> Hello, I voluntarily disconnected the cable to test the plugion. The printer returned home but when I ran the recovery file, the print went back too high. Is there a way to adjust the pickup pitch or just the Z-hop pitch when home? I don't use Klipper for information
> 
> Can we measure the value of going up from Z during Home XY and enter it in the plugin?

Yes, this is the value you must enter for Z_HOMING_HEIGHT

> I see in the Gcode this
>
> ; Klipper Z is not checked ; Z enable is not checked
>
> G90 ;absolute positioning G28 X0 Y0 ;home X/Y to min endstops G92 Z3.1 ;set Z with any homing offsets ;M211 S0 ; [...]

camarade35, 2023-02-02:
Thx for reply but I'm maybe stupid, I don't see where I can set this.
Thx for Help

paukstelis, 2023-02-02:
[image]

ackerthehacker2, 2023-02-03:
What do you set Z homeing height to when G28 X0 Y0  doesn't change the Z axis (Prusa)?  Leave it at 0?

paukstelis, 2023-02-03:
> What do you set Z homeing height to when G28 X0 Y0 doesn't change the Z axis (Prusa)? Leave it at 0?

Yes, leave it at 0.

camarade35, 2023-02-03:
really thanks and sorry for my inexperience

plundrigan, 2023-02-23:
I am using Marlin 2.1.1

If Z_HOMING_HEIGHT is commented out, but  Z_AFTER_HOMING = 10

Would I enter 10 in this Z Homing height?

Current value is 0 and had a failure and tried recovery, printed in mid air. Found this thread that seems applicable.

paukstelis, 2023-02-23:
> I am using Marlin 2.1.1
> 
> If Z_HOMING_HEIGHT is commented out, but Z_AFTER_HOMING = 10
> 
> Would I enter 10 in this Z Homing height?
> 
> Current value is 0 and had a failure and tried recovery, printed in mid air. Found this thread, thought this setting might be applicable.

I'm not entirely sure why it would raise if Z_HOMING_HEIGHT is commented out unless something has changed Marlin. The Z_AFTER_HOMING is the distance to set the Z axis after it has been homed. Since the plugin never homes the Z axis that shouldn't have an impact. You can follow the instructions in the readme and send a `G28 X Y` after you disconnect/connect the printer and see what the Z-axis does.

plundrigan, 2023-02-23:
"Since the plugin never homes the Z axis that shouldn't have an impact. "

When I ran the recovery print file. the print head did move back to the left and touch the x sensor twice. Then that is when it went back to the print and printed in the air. I did not notice when it moved up.

paukstelis, 2023-02-23:
It homes X and Y. It does not home Z.

plundrigan, 2023-02-23:
Is that a question or statement. :) I will test the gcode you suggested to see what happens with a disconnect.Sent from my Verizon, Samsung Galaxy smartphoneGet Outlook for AndroidFrom: paukstelis ***@***.***>Sent: Thursday, February 23, 2023, 4:13 PMTo: pablogventura/OctoPrint-PowerFailure ***@***.***>Cc: AmazinPistolPete ***@***.***>; Comment ***@***.***>Subject: Re: [pablogventura/OctoPrint-PowerFailure] Z height correction for recovery (Issue #12)
It homes X and Y. It does not home Z.

plundrigan, 2023-02-28:
I think I might know how my situation is different. I re-read one of your comments. The plugin only performs a G28 X Y upon disconnect/connect.  I had a M112 which forced the machine to be turned on and off. Since this is seen like a startup. I wonder if that is what is causing the z adjustment. Bear with me I can still learning Marlin and the printer and how each feature behaves. 
 
 I had another failure M112. rebooted machine. Started the recovery job and x raised.  I then stopped the job, lowered the z axis manually, and got close. The started the recovery job again, x and y did home no change to z. So there is something at a cold startup that is adjusting the z axis and throwing off the recovery.  Still not sure what setting that is.  I do have a CR-Touch so maybe some combination of stuff.  Will keep tinkering until I fully understand what is happening.

paukstelis, 2023-03-13:
This is a complication that might not be easy to bypass. There is really no way of OctoPrint knowing if the printer has been restarted, since it will look just like a disconnect. The only possibility I can think of is a pop-up that asks if the printer has been restarted.

cakazz54, 2023-04-01:
Hello, MoveZ value is max 250, I want this value to be max 400. Is there a gcode for this? Please help. Ender 3 V2 4.2.7