pablogventura/OctoPrint-PowerFailure #18: print head moved up 10mm while resuming with bltouch.

anand-010, 2023-02-15:
Before I've installed bltouch everything worked perfectly after I've installed it the head moved 10mm up before resuming and print in the air.
I think safe_z_home is the problem not sure I've used in 10mm many places

```
[stepper_x]
step_pin: gpio11
dir_pin: gpio10
enable_pin: !gpio12
microsteps: 16
rotation_distance: 40
#endstop_pin: ^gpio4
endstop_pin: tmc2209_stepper_x:virtual_endstop
homing_retract_dist: 0
position_endstop: 0
position_max: 200
homing_speed: 50

[tmc2209 stepper_x]
uart_pin: gpio9
tx_pin: gpio8
uart_address: 0
run_current: 0.8
hold_current: 0.6
stealthchop_threshold: 999999
diag_pin: ^gpio4
driver_SGTHRS: 100

[stepper_y]
step_pin: gpio6
dir_pin: !gpio5
enable_pin: !gpio7
microsteps: 16
rotation_distance: 40
endstop_pin: ^gpio3
# endstop_pin: tmc2209_stepper_y:virtual_endstop
# homing_retract_dist: 0
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_y]
uart_pin: gpio9
tx_pin: gpio8
uart_address: 2
run_current: 0.8
hold_current: 0.6
stealthchop_threshold: 999999
diag_pin: ^gpio3
driver_SGTHRS: 100

[stepper_z]
step_pin: gpio19
dir_pin: !gpio28
enable_pin: !gpio2
microsteps: 16
rotation_distance: 4
#endstop_pin: ^gpio25
#position_endstop: 0
endstop_pin: probe:z_virtual_endstop
position_min: -6.0
position_max: 250
homing_speed: 10

[tmc2209 stepper_z]
uart_pin: gpio9
tx_pin: gpio8
uart_address: 1
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 999999

[extruder]
step_pin: gpio14
dir_pin: !gpio13
enable_pin: !gpio15
microsteps: 16
#rotation_distance: 7.5
rotation_distance: 7.64
#nozzle_diameter: 0.2
nozzle_diameter: 0.4
filament_diameter: 1.75
#pressure_advance: 0.1
heater_pin: gpio23
sensor_type: EPCOS 100K B57560G104F
sensor_pin: gpio27
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 300
max_extrude_cross_section:2

[tmc2209 extruder]
uart_pin: gpio9
tx_pin: gpio8
uart_address: 3
run_current: 0.7
hold_current: 0.6
stealthchop_threshold: 999999

[filament_switch_sensor runout_sensor]
switch_pin: ^!gpio16
pause_on_runout: True

# [filament_motion_sensor smart_sensor]
# switch_pin: ^gpio16
# detection_length: 2.5

[heater_bed]
heater_pin: gpio21
sensor_type: EPCOS 100K B57560G104F
sensor_pin: gpio26
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: gpio17

[heater_fan hotend_fan]
pin: gpio18
heater: extruder
heater_temp: 50.0

[heater_fan controller_fan]
pin: gpio20
heater: heater_bed
heater_temp: 45.0

[mcu]
#serial: /dev/serial/by-id/usb-Klipper_rp2040_4550357129105258-if00
serial: /dev/ttyAMA0
restart_method: command

[printer]
kinematics: cartesian
max_velocity: 60
max_accel: 3000
max_z_velocity: 10
max_z_accel: 30

[bltouch]
#stow_on_each_sample: False
#probe_with_touch_mode: False 
sensor_pin: gpio22
control_pin: gpio29
#pin_move_time: 0.5
#pin_up_touch_mode_reports_triggered: False
x_offset: -52.8
y_offset: -7.2
#z_offset: 4.520
z_offset: 2.6
speed: 10
lift_speed: 20
samples: 2
sample_retract_dist: 10
samples_tolerance_retries: 2

[bed_mesh]
speed: 80
horizontal_move_z: 10
#mesh_min: 34, 9
#mesh_max: 199, 218
mesh_min: -18.8, -1.8
mesh_max: 147.2, 210.8
probe_count: 5, 5
mesh_pps: 2,2
fade_start: 1
fade_end: 10
fade_target: 0

[safe_z_home]
home_xy_position: 100, 100 # Change coordinates to the center of your print bed
speed: 50
z_hop: 10                 # Move up 10mm
z_hop_speed: 5

[neopixel board_rgb]
pin: gpio24
chain_count: 1
color_order: GRB
initial_RED: 0.3
initial_GREEN: 0.3
initial_BLUE: 0.3

[input_shaper]
shaper_freq_x: 75.2
shaper_type_x: mzv
shaper_freq_y: 76.2
shaper_type_y: 2hump_ei

[printer]
max_accel: 3000  # should not exceed the estimated max_accel for X and Y axes

[force_move]
enable_force_move: True

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	  0.015000, -0.090000, -0.182500, -0.236875, -0.263125
#*# 	  0.140625, -0.039375, -0.187500, -0.288750, -0.373750
#*# 	  0.349375, 0.114375, -0.073750, -0.226250, -0.372500
#*# 	  0.629375, 0.315000, 0.088750, -0.122500, -0.277500
#*# 	  1.003750, 0.552500, 0.273750, 0.010000, -0.229375
#*# tension = 0.2
#*# min_x = -18.8
#*# algo = lagrange
#*# y_count = 5
#*# mesh_y_pps = 2
#*# min_y = -1.8
#*# x_count = 5
#*# max_y = 210.8
#*# mesh_x_pps = 2
#*# max_x = 147.2
```

paukstelis, 2023-02-15:
This is directly from the README:
Klipper firmware. You must have the [force_move] section with the enable_force_move=true option in your Klipper configuration. Check the appropriate box in the settings. If [safe_z_home] is set, use the z_hop value as Z_HOMING_HEIGHT.

paukstelis, 2023-02-19:
You have to set the Z_HOMING_HEIGHT in the settings. If it raises 10mm when you do G28 XY, then set that value to 10 in the settings.

closed 2026-04-07 by anand-010