pablogventura/OctoPrint-PowerFailure #20: Does this remember the last location?

csm10495, 2023-03-28:
If I lose power and then move the printer's XYZ, then upon trying to continue the print, will it automatically go to the last XYZ before continuing?

paukstelis, 2023-03-28:
Yes. But it will rehome the printer first.

csm10495, 2023-03-28:
How does that work if there is already a part there? Won't it hit into it?

csm10495, 2023-03-29:
Doesn't rehoming bring it down to Z=0 though? Wouldn't that hit the part? 

(Sorry if these are silly questions)

paukstelis, 2023-03-29:
> Doesn't rehoming bring it down to Z=0 though? Wouldn't that hit the part?
> 
> (Sorry if these are silly questions)

No worries. Install the plugin and take a look. It homes X and Y, not Z. Z is set to whatever the Z value was when it crashed.

csm10495, 2023-03-29:
Ahhh that makes sense. So as long as Z doesn't move, it has the best shot of continuing. Thanks!