pablogventura/OctoPrint-PowerFailure #27: Bugs in wizard

thomergil, 2023-06-06:
Thanks for working on this.

One, there is no Next button, which the text refers to. There is only a Finish button. Secondly, the link to github is broken.

thomergil, 2023-06-06:
It looks like the <a> surrounds the buttons, making it impossible to click on the Finish button without going to the broken link.

thomergil, 2023-06-06:
You should fix this immediately.

thomergil, 2023-06-06:
...even when the missing " is fixed, the Finish button doesn't do anything.

paukstelis, 2023-06-06:
It has already been fixed. Reinstall the plugin.

thomergil, 2023-06-06:
I can't. Because the wizard blocks access to anything else.

paukstelis, 2023-06-06:
append `recovery` to your url, e.g.  http://octopi.local/recovery
Remove the plugin and reinstall.

RicSkye, 2023-06-06:
How do I remove the plugin in safe mode?

dioimnc, 2023-06-08:
When I'm in the recovery menu it won't switch to safe mode, saying "You lack admin permissions."
Im the only user so I should be admin right? What to do?

paukstelis, 2023-06-08:
You can also just delete the plugin.
Assuming you are using OctoPi, ssh into the Pi:
`rm -rf ~/oprint/lib/python3.7/site-packages/octoprint_powerfailure`
Depending on your OctoPi version that `python3.7` might be something else, in which case you would need to change accordingly.

dioimnc, 2023-06-08:
Okay, that worked out well. Thanks for the quick support, I appreciate it.

paukstelis, 2023-06-08:
> Okay, that worked out well. Thanks for the quick support, I appreciate it.

You can reinstall the plugin and it will be fixed from now on.

dioimnc, 2023-06-08:
> 

I did and everything worked fine this time. Thanks again.

closed 2023-06-28 by paukstelis