pablogventura/OctoPrint-PowerFailure #30: Poor documentation

thomergil, 2023-06-11:
Thanks for the plugin. However, it is unclear what a user is supposed to do after a power failure. What steps to resume the print? Thank you.

paukstelis, 2023-06-28:
Connect your printer then print the recovery file. Pretty straightforward.

Hairyloon, 2023-07-03:
> Connect your printer then print the recovery file. Pretty straightforward.

Someone with an ounce of smarts would put that into the "Readme" file.  
Someone with a little more than an ounce would also suggest where the recovery file is likely to be found.

paukstelis, 2023-07-03:
> > Connect your printer then print the recovery file. Pretty straightforward.
> 
> Someone with an ounce of smarts would put that into the "Readme" file. Someone with a little more than an ounce would also suggest where the recovery file is likely to be found.

Someone with the ability to read would see that the first bullet point of the README says: `By default, when there is a power failure the plugin generates the gcode and selects the recovery file once the printer is reconnected.`

thomergil, 2023-07-03:
Right. This has never worked for me. There is no recovery file. Where would it be? Does this require the Pi to be on a UPS? When my power goes out, both the printer and the Pi go out. If there are underlying assumptions in the setup, you should explain them.

The fact that people don't understand your otherwise cool plugin is not necessarily a sign of them being stupid. It may just be that they don't know the assumptions.

paukstelis, 2023-07-03:
> Right. This has never worked for me. There is no recovery file. Where would it be? Does this require the Pi to be on a UPS? When my power goes out, both the printer and the Pi go out. If there are [...]

As soon as OctoPrint is restarted and the printer is reconnected a recovery gcode file is generated and automatically selected. All you should have to do is click the print button. If you don't have file(s) called `recovery_<originalfilename>.gcode` then there is some other issue.

thomergil, 2023-07-03:
Thank you, @paukstelis. I don't think this has ever worked for me, and there is no recovery_* file. But I'll simulate it later by switching off the power strip that powers both the Pi and the printer. Thank you for now.

paukstelis, 2023-07-03:
If that is the case, then turn on logging for the plugin and provide the octoprint.log file.

thomergil, 2023-07-03:
Cool. Will do.

thomergil, 2023-07-03:
Out of curiosity -- how does this work? How does the plugin know on restart what was being printed and how far it got?

paukstelis, 2023-07-03:
> Out of curiosity -- how does this work? How does the plugin know on restart what was being printed and how far it got? Does it log something and it re-reads its own log? (I'm too lazy to check the source code. 😄 )

It writes out the position in the printing file to a json file every second (or at a user determined frequency) along with other relevant information. Then it just makes a new file (the recovery_<filename>.gcode) with the plugin-derived start gcode.