pablogventura/OctoPrint-PowerFailure #32: Can this also recover from an emergency stop?

Hairyloon, 2023-07-01:
It seems to me obvious that the plugin ought to be able to recover from an emergency stop, or some other kind of printing failure, but if it is only triggered by a power failure, then it isn't going to.
Of course you probably don't want to automatically resume after an emergency stop, because you need to deal with the emergency, but it would be handy to be able to run it when you're ready.
Can this be done?

paukstelis, 2023-07-02:
Not without a lot of work. Emergency stop will effectively cancel the print. As of right now, cancelled prints aren't setup so they can be recovered. Quite a lot would have to change and it is not something I am particularly keen on doing.

Hairyloon, 2023-07-02:
> Not without a lot of work. Emergency stop will effectively cancel the print...

Does shutting the power off not cancel the print?
it always stops when it's happened to me...

paukstelis, 2023-07-02:
> > Not without a lot of work. Emergency stop will effectively cancel the print...
> 
> Does shutting the power off not cancel the print? it always stops when it's happened to me...

No. It terminates the print, sure, but that is because the server has died. Cancelling a print means the user has chosen to cancel the print by hitting a button.

Hairyloon, 2023-07-02:
> No. It terminates the print, sure, but that is because the server has died. Cancelling a print means the user has chosen to cancel the print by hitting a button.

And the significant difference is...

paukstelis, 2023-07-02:
One is user initiated by clicking the cancel button. One is accidental because the power went off.

Hairyloon, 2023-07-02:
OK,, so with one, we can have the computer do almost anything we want it to after the button is pressed, whereas with the other, it just goes down and if we're lucky it doesn't corrupt any data...
Apparently the latter one is easier to recover from...

paukstelis, 2023-07-03:
> OK,, so with one, we can have the computer do almost anything we want it to after the button is pressed, whereas with the other, it just goes down and if we're lucky it doesn't corrupt any data... Apparently the latter one is easier to recover from...

No, but what is easier is knowing the intention. If the power goes out or the printer disconnects during a print, that wasn't the intention of the operator. If the print is cancelled, as you suggest, it could have been cancelled because of an e-stop, as you seem to want, or it could have been cancelled because the operator selected the wrong file, or the operator realized they had sliced for the wrong filament, etc. etc. Therefore, the plugin would have to determine the intention of the user after every cancellation with a pop-up or something saying "Hey this print was cancelled, do you want to recover from the previous cancellation?" which would be both annoying and a pain to have to incorporate. 

Basically, it comes down to: 1. I won't be including that feature. 2. If you want it, write it, and I (or someone else) will review a PR and decide if it fits.

closed 2023-07-03 by paukstelis

paukstelis, 2023-07-03:
As I say, you are very welcome to write it yourself and submit a PR.