pablogventura/OctoPrint-PowerFailure #33: How is this used? More documentation needed

Charmandrigo, 2023-07-16:
I often get random disconnects in the middle of my prints, so I assumed this would help me but, even when I have the addon enabled nothing really happens on reconnections or even if simulate an outage to test, there's no special button to restore from the last saved gcode, nor I see any recovered gcode on my files list, despite I have automatic restart enabled and save frequency set to 1s

paukstelis, 2023-07-16:
If you are getting disconnects often, you are best off figuring out why that is and fix the problem

Take a look at your log file and see what it says after reconnection. You can search for `powerfailure` to find messages from the plugin.

Charmandrigo, 2023-07-16:
there's no log file, I have to enable it manually and see what it outputs.
I should add that my printer currently has SDSUPPORT disabled because my motherboard and display don't have an sd slot so I print everything through serial connection and the octoprint internal storage, this does not require marlin to have any specific feature enabled? Is there any other plugin that may break this one?

paukstelis, 2023-07-16:
just the octoprint.log file will have the info.
It does not require anything special from Marlin.

Charmandrigo, 2023-07-16:
This is the latest file it saved

[octoprint.log](https://github.com/pablogventura/OctoPrint-PowerFailure/files/12065006/octoprint.log)

paukstelis, 2023-07-16:
Looks like it is an issue with a plugin, SlicerEstimator, or perhaps DisplayLayerProgress.
Try disabling and see what happens.

```2023-07-16 11:09:42,701 - octoprint.plugins.powerfailure - INFO - Recovering from a print failure
2023-07-16 11:09:42,711 - octoprint.plugins.octoeverywhere - INFO - Connected To OctoEverywhere, server con wss://sfo.octoeverywhere.com/octoclientws[1]. Starting handshake...
2023-07-16 11:09:42,738 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 2.1.2 (Jan  4 2023 20:29:23)"
2023-07-16 11:09:42,819 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2023-07-16 11:09:42,923 - octoprint.plugins.camerasettings - WARNING - Error setting white_balance_temperature:
white_balance_temperature: Input/output error
VIDIOC_QUERYCAP: ok
VIDIOC_S_CTRL: failed: Input/output error

2023-07-16 11:09:42,937 - octoprint.util.comm - INFO - Firmware states that it supports emergency GCODEs to be sent without waiting for an acknowledgement first
2023-07-16 11:09:43,303 - octoprint.plugins.tracking - INFO - Sent tracking event pong, payload: {'version': '1.9.1', 'os': 'linux', 'bits': 32, 'python': '3.7.3', 'pip': '20.3.3', 'cores': 4, 'freq': 1500.0, 'ram': 1909071872, 'pi_model': 'Raspberry Pi 4 Model B Rev 1.2', 'octopi_version': '0.18.0', 'octopiuptodate_build': '0.18.0-1.7.3-20220120112925', 'plugins': 'touchui:0.3.18,themeify:1.2.2,rtmpstreamer:1.0.4,powerfailure:1.2.1,smartabl:1.2.2,slicerestimator:1.6.2,simpleemergencystop:1.0.5,octopod:0.3.15,multicam:0.4.0,marlinslider:0.2.1,heatertimeout:0.0.4,excluderegion:0.3.2,camerasettings:0.4.3,octoeverywhere:2.7.0,displaylayerprogress:1.28.0,change_filament:0.3.2,buffer_buddy:0.1.0,bedlevelvisualizer:1.1.1,actioncommands:0.5'}
2023-07-16 11:14:11,644 - octoprint.filemanager - ERROR - Error when calling preprocessor hook for plugin SlicerEstimator, ignoring
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/__init__.py", line 708, in add_file
    allow_overwrite=allow_overwrite,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SlicerEstimator/__init__.py", line 181, in on_file_upload
    filedata = SlicerEstimatorFiledata(path, file_object, self)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SlicerEstimator/metadata.py", line 120, in __init__
    self.slicer = SlicerEstimatorMetadataFiles.detect_slicer(self._file_object.path)
AttributeError: 'StreamWrapper' object has no attribute 'path'
2023-07-16 11:14:11,645 - octoprint.plugins.DisplayLayerProgress - INFO - FilePreProcessor. Checking LayerExpressions.
2023-07-16 11:14:11,646 - octoprint.plugins.DisplayLayerProgress - INFO - FilePreProcessor. LayerExpression valid. Start processing...
2023-07-16 11:14:11,974 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'http://192.168.31.138/', 'path': '/', 'query_string': 'l10n=en'}
2023-07-16 11:14:12,017 - octoprint.plugin - ERROR - Error while calling plugin powerfailure```

Charmandrigo, 2023-07-17:
`2023-07-16 10:01:33,432 - octoprint.plugin - ERROR - Error while calling plugin powerfailure
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 275, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_powerfailure/__init__.py", line 296, in on_event
    self.check_recovery()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_powerfailure/__init__.py", line 130, in check_recovery
    recovery_fn = self.generateContinuation()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_powerfailure/__init__.py", line 211, in generateContinuation
    octoprint.filemanager.FileDestinations.LOCAL, recovery_fn, stream, allow_overwrite=True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/__init__.py", line 733, in add_file
    display=display,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/storage.py", line 894, in add_file
    file_object.save(file_path)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/util.py", line 102, in save
    shutil.copyfileobj(source, dest)
  File "/home/pi/oprint/lib/python3.7/shutil.py", line 79, in copyfileobj
    buf = fsrc.read(length)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/util.py", line 211, in read
    line = self.input_stream.readline()`

What does this mean? I am still seeing this error after uninstalling and disabling a lot of my addons, is this addon meant to show a message or a warning on the Octoprint UI?

paukstelis, 2023-07-17:
No, it is an error, but the cause is unknown. Either some plugin is messing with the files, or perhaps you are out of room on the Pi? Did you disable that Slicer Estimates plugin?

Charmandrigo, 2023-07-17:
[image]
I don't think storage is a problem.

Yeah, I disabled Slicer Estimator and displaylayerprogress

Charmandrigo, 2023-07-17:
[octoprint(1).log](https://github.com/pablogventura/OctoPrint-PowerFailure/files/12065811/octoprint.1.log)
There are new logs after disabling more, I don't know if the last lines on the log indicates the plugin is doing it's job now

paukstelis, 2023-07-17:
I don't know which one breaks it, but you seem to have some plugins that aren't in the repository (bufferbuddy) which would be the first suspects.

Charmandrigo, 2023-07-17:
Bufferbuddy was still enabled when the plugin started to work, so it can't be that one.
I only disabled Slicerestimate, display layer progress, themeify and camera settings, and uninstalled touchui

paukstelis, 2023-07-17:
From the logs it looks like it was Slicer Estimate. If you look at line 17589, it recovers from the failure (no errors following). Scroll up to your plugin list and Slicer Estimate was disabled.

closed 2023-09-24 by paukstelis