pablogventura/OctoPrint-PowerFailure #35: What do I set for my Z homing height?

pto199, 2023-12-02:
I did the G28 command and got this:

Send: G28 X0 Y0
Recv: echo:busy: processing
Recv: X:-10.00 Y:-8.00 Z:5.00 E:0.00 Count X:-800 Y:-640 Z:2000
Recv: ok P15 B3

Whats does that mean?  Is my Z homing height supposed to be set as 5?  as 2000?  as something else?  The instructions are not really clear.

paukstelis, 2023-12-02:
You should set Z homing height to 5.

paukstelis, 2024-11-10:
> I agree. Reading "... Critical: Determine if your printer has Z_HOMING_HEIGHT set. This setting raises the Z-axis on any homing event to avoid collisions. You can check your printer firmware configuration or in a resting state issue the command G28 X0 Y0 in the command terminal and observe if the Z-axis is raised, and by how much. This value is used for Z_HOMING_HEIGHT...." It is not clear to me what the desired situation is? Is having a value for Z_HOMING_HEIGHT a good thing? Or a bad thing to avoid?

It is a necessary thing, depending on your printer's firmware.