pablogventura/OctoPrint-PowerFailure #37: Z homing problems

Toscarlee, 2024-03-17:
Hello, greetings everyone, the problem I have is the following
I use the command g28 x0 y0 it tells me that I have a 10mm z homing
When the printer gives me the recovery file, the first thing it does is go up 10 mm, it goes to home and then it goes down 10 mm and then it goes to z0 and tries to restart the print attached to the bed as if it were a new print.

paukstelis, 2024-03-17:
Upload your recovery gcode file and a screenshot of your PowerFailure settings.

Toscarlee, 2024-03-18:
Recv:  T:210.21 /210.00 B:59.88 /60.00 @:65 B@:34
Recv: X:-9.00 Y:-6.00 Z:10.00 E:11.10 Count X:-720 Y:-480 Z:4000
Recv: ok
Send: N9 G92 Z12.8*116
Recv: X:-9.00 Y:-6.00 Z:12.80 E:11.10 Count X:-720 Y:-480 Z:4000
Recv: ok
Send: N10 G91*32
Recv: ok
Send: N11 G1 Z-10.0 F200*4
Recv: ok
Send: N12 G90*35
Recv: ok
Send: N13 M83*42
Recv: ok
Send: N14 G1 E3.0 F100*2
Recv: ok
Send: N15 G92 E0*115
Recv: X:-9.00 Y:-6.00 Z:2.80 E:0.00 Count X:-720 Y:-480 Z:4000
Recv: ok
Send: N16 M82*46
Recv: ok
Send: N17 G1 X106.645 Y113.851 F3000.0*107
Recv:  T:210.16 /210.00 B:59.88 /60.00 @:66 B@:35
Recv: echo:busy: processing
Recv:  T:210.18 /210.00 B:59.90 /60.00 @:65 B@:36
Recv: echo:busy: processing
Recv:  T:210.14 /210.00 B:60.01 /60.00 @:65 B@:30
Recv: echo:busy: processing
Recv: ok
Send: N18 M106 S255*92
Recv: ok
Send: N19 G0 F776.5*90
Recv: ok
Send: N20 G92 E11.09661*83
Recv: X:106.64 Y:113.85 Z:2.80 E:11.10 Count X:4619 Y:5052 Z:-201
Recv: ok
Send: N21 G1 X105.626 Y112.637 E11.03553*103
Recv: ok
Send: N22 G1 X105.845 Y112.97 E11.05083*93
Recv: ok
Send: N23 G1 X106.118 Y113.32 E11.06787*80
Recv: ok
Send: N24 G1 X106.353 Y113.577 E11.08124*111
Recv: ok
Send: N25 G1 X106.645 Y113.851 E11.09661*98
Recv: ok
Send: N26 G1 X106.95 Y114.098 E11.11168*87
Recv: ok
Send: N27 G1 X107.276 Y114.321 E11.12684*111
Recv:  T:210.08 /210.00 B:60.03 /60.00 @:65 B@:31
Recv: ok
Send: N28 G1 X107.62 Y114.519 E11.14208*92
Recv: ok
Send: N29 G1 X107.975 Y114.689 E11.15719*108
Recv: ok
Send: N30 G1 X108.347 Y114.833 E11.17251*100
Recv: ok
Send: N31 G1 X108.728 Y114.947 E11.18778*107
Recv: ok
Send: N32 G1 X109.113 Y115.031 E11.20291*103
Recv: ok
Send: N33 G1 X109.508 Y115.084 E11.21821*102
Recv: ok
Send: N34 G1 X109.903 Y115.108 E11.2334*93
Recv: ok
Send: N35 G1 X110.296 Y115.1 E11.24849*110
Recv: ok
Send: N36 G1 X110.692 Y115.061 E11.26377*111
Recv: ok
Send: N37 G1 X111.084 Y114.993 E11.27904*100
[... 260 lines cut ...]
Recv:  T:188.66 /0.00 B:58.98 /0.00 @:0 B@:0
Recv:  T:186.17 /0.00 B:58.85 /0.00 @:0 B@:0
Recv:  T:183.68 /0.00 B:58.79 /0.00 @:0 B@:0
Recv:  T:181.21 /0.00 B:58.63 /0.00 @:0 B@:0
Recv:  T:178.92 /0.00 B:58.60 /0.00 @:0 B@:0
Recv:  T:176.71 /0.00 B:58.50 /0.00 @:0 B@:0
Recv:  T:174.45 /0.00 B:58.39 /0.00 @:0 B@:0
Recv:  T:172.31 /0.00 B:58.33 /0.00 @:0 B@:0
Recv:  T:170.11 /0.00 B:58.23 /0.00 @:0 B@:0
Recv:  T:168.12 /0.00 B:58.14 /0.00 @:0 B@:0
Recv:  T:166.07 /0.00 B:58.06 /0.00 @:0 B@:0
Recv:  T:164.09 /0.00 B:57.99 /0.00 @:0 B@:0
Recv:  T:162.09 /0.00 B:57.87 /0.00 @:0 B@:0
Recv:  T:160.09 /0.00 B:57.77 /0.00 @:0 B@:0
Recv:  T:158.24 /0.00 B:57.66 /0.00 @:0 B@:0
Recv:  T:156.38 /0.00 B:57.57 /0.00 @:0 B@:0
Recv:  T:154.52 /0.00 B:57.47 /0.00 @:0 B@:0
Recv:  T:152.79 /0.00 B:57.35 /0.00 @:0 B@:0
Recv:  T:151.03 /0.00 B:57.31 /0.00 @:0 B@:0
Recv:  T:149.30 /0.00 B:57.18 /0.00 @:0 B@:0
Recv:  T:147.65 /0.00 B:57.09 /0.00 @:0 B@:0
Recv:  T:146.00 /0.00 B:56.99 /0.00 @:0 B@:0
Recv:  T:144.35 /0.00 B:56.88 /0.00 @:0 B@:0
[image]
[image]

Toscarlee, 2024-03-18:
[image]
[image]
[recovery_CILINDRO PRUEBA.txt](https://github.com/pablogventura/OctoPrint-PowerFailure/files/14635592/recovery_CILINDRO.PRUEBA.txt)

paukstelis, 2024-03-18:
Everything looks correct:
```
G28 X0 Y0 ;home X/Y to min endstops
G92 Z12.8 ;set Z with any homing offsets
;M211 S0 ; Deactive software endstops
G91 ;relative positioning
G1 Z-10.0 F200 ; correcting Z_HOMING_HEIGHT
G90 ;absolute positioning
```

G92 Z12.8 sets the Z-height with the homing offfset. This means your printer failed at Z=2.8mm
No where in the gcode does it go back to Z=0.