pablogventura/OctoPrint-PowerFailure #38: Resume X offset issue

Digicrat, 2024-05-13:
I've (literally) dusted off my old CTC Formaker (a Makerbot Replicator Dual clone with extras, still running the original firmware) in the past week.  It worked great for the past few days, but the Octoprint connection keeps disconnecting with an I/O error every few hours today, so I installed this plugin (while also trying to debug why that's happening).

The plugin seems to work when I reconnect the printer and hit restart, except for the fact that it's off in at least the X axis (I can't tell from the few seconds I let it run if the Y axis is also off).  It's probably off by about an inch and a half towards the home position.  There are plenty of notes concerning Z axis offset issues, but XY offset seems to be a new one.

Note: I don't think it's related, but (in case it is) I have previously seen and suppressed the "Exceeds the print volume of the currently selected printer profile" warning in Octoprint after failing to solve it with a 'Custom Bounding Box'.  The prints do appear centered properly in the Gcode Viewer though.

Any ideas?  I've left it on the build plate for now .

paukstelis, 2024-05-13:
You may need to modify the recovery start gcode. It should home X and Y axes, but if you are on original firmware that is probably Sailfish, which is not going to use the same commands as Marlin.

Digicrat, 2024-05-14:
That makes sense, thanks.  This looks like it might take some thought to figure out exactly what's needed/missing on my end ... especially since I'm not very knowledgeable of the gcode usage yet.  A few substitutions look obvious on first glance at just the XY homing based on the comments (G161+G162 in place of G28 for homing), but there's a number of other lines interspersed that I'm not as sure about.

For reference, I'm using a PrusaSlicer config that includes the start gcode taken from https://github.com/DrLex0/FFCP-GCodeSnippets/blob/master/GCode/Start-right-extruder.gcode