pablogventura/OctoPrint-PowerFailure #41: Snapmaker 2 A350 with octoprint.  Want to understand how to make it work

chinloon, 2025-06-11:
I have been trying to figure out how to make the plug-in work.  I could see the .json file being updated as printing is progressing.  However, when printer failed, I simulated by powering off the printer (octoprint was not powered off) and then restarted the printed.  Then I manually connect to printer on octoprint.  I don't see any any recovery_** file created from the failed print.  Is there something I need to watch out for?

paukstelis, 2025-06-11:
Just turning off the printer may not be sufficient. Many printer main boards can also receive power through USB. If the main board is still powered, it doesn't know that the printing has stopped. A better test is to just pull out the USB cable during printing.

chinloon, 2025-06-12:
Hmm... Thought octoprint recovery plug-in also detects for connection failure (as that causes printing to stop).  Hos is that detection one?  Restoration of serial connection?  Was thinking if so, then when I click the connect to reconnect to printer after a mid-way print job, a recovery gcode is created?